ivocavalcante/PrusaSlicer-snap #1: Theming is missing

Cediddi, 2020-11-05:
I've just started using this build. From what I see, theming is missing. How can I help you about this issue?

[image]

ivocavalcante, 2020-11-05:
Hi @Cediddi , thanks for reporting.

Could you please provide more info? Linux distribution, snap version, theme you're using, for instance.

ivocavalcante, 2020-11-07:
I uploaded a new version; can you try it, please?

ivocavalcante, 2020-11-15:
Hello @Cediddi , any feedback on this?

Cediddi, 2020-11-17:
I'm sorry for late reply.

> Linux distribution, snap version, theme you're using, for instance.
snap    2.47.1+18.04
snapd   2.47.1+18.04
series  16
ubuntu  18.04
kernel  5.4.0-48-generic
plasmashell 5.12.9

Theme: Materia Dark [KDE], Breeze-Dark [GTK2/3]

Cediddi, 2020-11-17:
Problem persists with edge branch + a new issue of ssl certs

[image]
[image]

ivocavalcante, 2020-11-21:
Ok @Cediddi , let's go:

The message about SSL is not an issue, it comes with 2.3.0 version of the app, even on the AppImage they build and provide themselves. It's just an information and most people would safely click "Remember my choice" and go on.

As for the theming, need some more info. Could you, please, provide the output of commands below?

`snap info prusa-slicer | grep installed:`

and

`snap connections prusa-slicer`

Cediddi, 2020-11-21:
For stable package, this are the outputs:
```
cediddi@aiur:~$ snap info prusa-slicer|grep installed:
installed:          2.2.0-snap2                   (13) 162MB -
cediddi@aiur:~$ snap connections prusa-slicer 
Interface               Plug                           Slot                              Notes
content[gtk-2-engines]  prusa-slicer:gtk-2-engines     gtk2-common-themes:gtk-2-engines  -
content[gtk-2-themes]   prusa-slicer:gtk-2-themes      gtk-common-themes:gtk-2-themes    -
content[icon-themes]    prusa-slicer:icon-themes       gtk-common-themes:icon-themes     -
content[sound-themes]   prusa-slicer:sound-themes      gtk-common-themes:sound-themes    -
desktop                 prusa-slicer:desktop           :desktop                          -
desktop-legacy          prusa-slicer:desktop-legacy    :desktop-legacy                   -
hardware-observe        prusa-slicer:hardware-observe  -                                 -
home                    prusa-slicer:home              :home                             -
mount-observe           prusa-slicer:mount-observe     -                                 -
network                 prusa-slicer:network           :network                          -
opengl                  prusa-slicer:opengl            :opengl                           -
optical-drive           prusa-slicer:optical-drive     :optical-drive                    -
raw-usb                 prusa-slicer:raw-usb           -                                 -
removable-media         prusa-slicer:removable-media   -                                 -
unity7                  prusa-slicer:unity7            :unity7                           -
wayland                 prusa-slicer:wayland           :wayland                          -
x11                     prusa-slicer:x11               :x11                              -
```

And for the edge version, outputs goes like this:
```
cediddi@aiur:~$ snap info prusa-slicer|grep installed:
installed:          2.3.0-alpha3-snap1            (16) 259MB -
cediddi@aiur:~$ snap connections prusa-slicer         
Interface               Plug                           Slot                              Notes
content[gtk-2-engines]  prusa-slicer:gtk-2-engines     gtk2-common-themes:gtk-2-engines  -
content[gtk-2-themes]   prusa-slicer:gtk-2-themes      gtk-common-themes:gtk-2-themes    -
content[icon-themes]    prusa-slicer:icon-themes       gtk-common-themes:icon-themes     -
content[sound-themes]   prusa-slicer:sound-themes      gtk-common-themes:sound-themes    -
desktop                 prusa-slicer:desktop           :desktop                          -
desktop-legacy          prusa-slicer:desktop-legacy    :desktop-legacy                   -
hardware-observe        prusa-slicer:hardware-observe  -                                 -
home                    prusa-slicer:home              :home                             -
mount-observe           prusa-slicer:mount-observe     -                                 -
network                 prusa-slicer:network           :network                          -
opengl                  prusa-slicer:opengl            :opengl                           -
optical-drive           prusa-slicer:optical-drive     :optical-drive                    -
raw-usb                 prusa-slicer:raw-usb           -                                 -
removable-media         prusa-slicer:removable-media   -                                 -
unity7                  prusa-slicer:unity7            :unity7                           -
wayland                 prusa-slicer:wayland           :wayland                          -
x11                     prusa-slicer:x11               :x11                              -
```

ivocavalcante, 2020-11-21:
Ok, installed Kubuntu 18.04 in a VM and reproduced the issue. I'll take a look at it.

ivocavalcante, 2020-11-22:
The problem is that GTK uses Xsettings protocol to define/query themes and other properties; KDE doesn't use it, resorting to the ~/.gtkrc-2.0 file (this is what you change when use KDE GTK interface), and strictly confined Snaps (as PrusaSlicer Snap) can't access these by default. I'll take the case to Snap forum, see if anyone has a solution but, for now, you can copy the file to snap home and it'll work:
```
cp ~/.gtkrc-2.0 ~/snap/prusa-slicer/current/
```

You'll have to do it everytime you change GTK definitions on KDE interface (linking doesn't work also, 'cause of confinement).
I'll let you know if/when I have a definitive solution.

Please, let me know if this _solves_ your problem for now.

ivocavalcante, 2020-11-29:
Hello @Cediddi ,

Just released new beta version, GTK 3 based (was using GTK 2 'till alpha3, due to bugs on the UI). Tested this version on Kubuntu 18.04 and it seems to get themes right. Could you try this one, please?

Cediddi, 2020-11-30:
Exceptional work, thank you. 
[image]

closed 2020-11-30 by Cediddi