ivocavalcante/PrusaSlicer-snap #5: PrusaSlicer hangs after ever g-code export

driveawedge, 2021-05-07:
Starting after the last snap update (identified as happening on May 2, 2021), PrusaSlicer now hangs after I export g-code. The g-code is always correctly exported but the app itself freezes and I have to kill it. I've tried waiting several minutes for it to unfreeze but it is clearly in some frozen state. The system I use is not connected to the printer and I haven't changed any USB devices in the past few weeks.

Here are the system particulars...

installed version:
prusa-slicer             2.3.1+snap2                 26     latest/stable    ivo-cavalcante

snapd version:
snapd/groovy-updates,groovy-security,now 2.48.3+20.10 amd64 [installed,automatic]

Ubuntu 20.10, kernel 5.8.0-50-generic

ivocavalcante, 2021-05-07:
Thanks for reporting, Dan. I was able to reproduce, but am still to find a cause. I'll report back as soon as I find something.

ivocavalcante, 2021-05-09:
Hello Dan,

I found it to be related to the size of output file path. More precisely, it is related to the routine that shows the popup when export is completed - it is not being able to handle width overflow. It doesn't happen upstream, I checked, so you can revert to using it, if you prefer. I'll continue checking the Snap package.

driveawedge, 2021-05-09:
Thanks Ivo!

ivocavalcante, 2021-08-22:
Hi @driveawedge , if you're still using this snap, you might want to try the version I uploaded on bet channel. If you do, please let me know if it fix the problem.

ivocavalcante, 2021-08-22:
Closed, same as #6 .

closed 2021-08-22 by ivocavalcante