ivocavalcante/PrusaSlicer-snap #6: Slicer crashes when clicking 'slice' button

peaprog, 2021-08-09:
Hello. This version of Prusa Slicer is great. It has been working well for me for over half  a year. However, all of a sudden, when I click the "Slice now" button, the slicer just crashes. I would appreciate if you could look into this. I have included output when the slicer is started via terminal.

[prusa_slicer_erros.txt](https://github.com/ivocavalcante/PrusaSlicer-snap/files/6951705/prusa_slicer_erros.txt)

Thanks!

ivocavalcante, 2021-08-09:
Sorry pal, I've been MIA for a month or two, for personal reasons. Will try to get back on track this weekend.

peaprog, 2021-08-10:
Thanks! I hope you can figure it out!

ivocavalcante, 2021-08-22:
Hi @peaprog , could you please try the version I uploaded on beta channel? Let'me know if it fix your problem.

closed 2021-08-22 by ivocavalcante

peaprog, 2021-08-22:
I'm really sorry, but where/how can I install from the beta channel.

ivocavalcante, 2021-08-22:
Sorry pal, uploaded to stable by mistake and, since some people had already updated it, decided to keep it there. If you refresh your Snaps, youl'll probably get it, no need to change channels.

peaprog, 2021-08-22:
Thanks, it works well now!

ivocavalcante, 2021-08-22:
Glad to hear, mate. Thanks for pointing out.