ivocavalcante/PrusaSlicer-snap #11: CLI Slicing Fails on Amazon Linux 2 with PrusaSlicer 2.6.0+snap3

JoJunYeong, 2023-07-18:
Hello,

I've encountered an issue when trying to use PrusaSlicer's CLI for slicing on Amazon Linux 2.

When using PrusaSlicer version 2.6.0+snap2, the CLI slicing works as expected. However, after updating to version 2.6.0+snap3, attempting to slice via the CLI results in a thread-related error.

Here are the steps to reproduce the issue:

Update PrusaSlicer to version 2.6.0+snap3 on Amazon Linux 2.
Attempt to slice a model using the CLI.
Expected Result:
The model should be sliced successfully, as it does with version 2.6.0+snap2.

Actual Result:
A thread-related error occurs, and the slicing process fails.

I would appreciate any guidance or fixes for this issue. Thank you for your time and assistance.

Best Regards
Jun

ivocavalcante, 2023-07-18:
Hi Jun,

For this release, I changed the base of the Snap to core22 and started
using Gnome extension, in order to reduce package size and improve
maintainability. Seems there are edges to iron out still.

Can you please send the output of:

`$ snap connections prusa-slicer`

Also, if you could include a snippet of the error message, might help.

JoJunYeong, 2023-07-21:
> 안녕하세요 Jun, 이번 릴리스에서는 패키지 크기를 줄이고 유지 관리를 개선하기 위해 Snap의 기반을 core22로 변경하고 Gnome 확장을 사용하기 시작했습니다. 여전히 다림질해야 할 가장자리가있는 것 같습니다. `$ snap connections prusa-slicer`의 출력을 보내주시겠습니까? 또한 오류 메시지의 스니펫을 포함할 수 있다면 도움이 될 것입니다. Em ter., 7월 18일 de 2023 06:44, Jun ***@***.***> escreveu:
> […](#)

this is the error log.
[2023-07-16 00:28:34.698023] [0x00007f7531a82c40] [trace]   Initializing StaticPrintConfigs
terminate called after throwing an instance of 'std::runtime_error'
  what():  pthread_create has failed: Operation not permitted

and this is result of snap connections prusa-slicer
[image]

thanks for your support!

ivocavalcante, 2023-08-02:
Hi,

Sorry for late reply, been away on the last two weeks. Could you please reproduce the problem an Ubuntu server? It should work on your environment, but I want to eliminate variables and since Ubuntu is a "definitely should work" distro, maybe we take some causes out.

JoJunYeong, 2023-08-03:
Hi, 

can i test it in ubuntu desktop?

ivocavalcante, 2023-08-03:
Yes, if you are able to reproduce your exact settings - profiles, model and all. Did some quick tests here, on Ubuntu desktop, and had no trouble, but since you probably use your own custom profiles and models, I recognize it's not a good test.

ivocavalcante, 2023-08-08:
Instead of compiling, you might try to revert to the revision that was working for you. Try:
`snap revert prusa-slicer --revision 87`

If it works, you can then _pin_ it, preventing further upgrades, with:
`snap refresh prusa-slicer --hold`

If these commands don't work as intended, tell here and I might be able to help you in compiling the package yourself. Nonetheless, I'd like you to help me investigate this issue further, if you're willing to. I need more info; if you could provide an exact model and profile that effectively fails on Amazon Linux, I may be able to reproduce. I noticed you used different models on your tests, so I'd need models and profiles that are known to cause failures.

ivocavalcante, 2023-08-11:
What command did you use in order to compile? It should be as simple as [installing](https://snapcraft.io/lxd) and [configuring](https://documentation.ubuntu.com/lxd/en/latest/tutorial/first_steps/#install-and-initialize-lxd) (I believe "Install and initialize LXD section is enough) `lxd`, and issuing:

`snapcraft pack --use-lxd --enable-manifest`

No need to run as root, just add your user to `lxd` system group and it's done. Remember to checkout the tag corresponding to the version you need.

Nonetheless, I suggest you update snapd version on your server, that's why you couldn't pin the revision.

`sudo snap refresh snapd`   (not sure it'll work, but worth a try)

JoJunYeong, 2023-08-12:
thank you :) 
sudo snap refresh snapd doesn't work in my environment.
I think snapd is fit at ubuntu, and my OS is similar with fedora, cent os 
So, last few days I'm trying to find another way instead using snapd.
And I find how to figure it out :)
Just download AppImage and execute. 
prusa slicer is worked!
and there is no error.
this comment for amazon linux user.
therefore, there is no need to compile snapcraft
I think @ivocavalcante  doesn't have to investigate this issue anymore.
I think it just depends on OS type of linux.

ivocavalcante, 2023-08-24:
I'm glad it worked JoJun. Support for Snap on RHEL-based systems is a little clunky, it has to be installed through a different repo than the official ones.If I find time, will try Amazon Linux myself, see if it works on a newer version of snapd.

closed 2023-08-24 by ivocavalcante